anduril/jetpack-nixos #5: Enable support for nvidia-container

danielfullmer, 2022-12-09:
We should be able to run docker/podman and systemd containers (OCI) on jetson devices using nvidia's container runtime support.

GTrunSec, 2023-03-07:
Any process here?

danielfullmer, 2023-03-07:
No--I'm not currently working on this.

GTrunSec, 2023-03-09:
I see, thx. Currently, because my friend took the jetpack device, I can't do full tests for the OCI.

j-baker, 2023-07-29:
I've spent a bit of time digging into this, and thus far have had success with errors.

- The built in Docker nVidia support mostly handles libnvidia-container1, nvidia-container-toolkit etc - not obviously much jetson-specific stuff in there. libnvidia-container0 is not present.
- It's not sufficient to just install libnvidia-container0. The way that libnvidia-container0 figures out how to mount things is with aid of the ldconfig cache, and so the traditional NixOS way (given there is no ldconfig cache) is to use /tmp/ld.so.cache. Applying similar patches helps here. I'm not actually sure any of this is relevant, given the below.
- At this point, we can launch containers. Unfortunately, no success with actually getting devices mounted. This is because Jetpack mounts binaries differently to normal nVidia because of all its library bundling - a CSV file at path `/etc/nvidia-container-runtime/host-files-for-container.d/l4t.csv` is written down in the root Jetson_Linux download, and this contains entries for each device, file, symlink, etc. The idea is that these then get remounted at the same path inside the container. There is no room for remapping paths.

For simplicity, I've attached the present `l4t.csv`. What I'm observing is that:

1. Devices are... obviously devices. They can stay as they are, they appear to mostly appear equally.
2. There are clearly some `.so` files which are provided by the video driver (e.g. libcuda.so). These presumably need some amount of special casing to ensure they end up in the container correctly. My hope however is that because 
3. There are a bunch of 'base libraries' which are auto mounted. An example might be gstreamer or libnvsci. As far as I can tell, most of these come from the list of debs already contained in this repository.

For my use case, it's critical that unmodified L4T containers run nicely.

At present therefore, I think a simple strategy would be:
1. Unpack the relevant debs into the nix store.
2. Build a new derivation containing paths from there, and the devices (so e.g. the list is `/dev/nvhost-gpu, ..., /nix/store/something-.../usr/lib/aarch64-linux-gnu/tegra/libnvscf.so, ...`). Build an appropriate CSV for this.
3. When building libnvidia-container0, replace the /etc/nvidia-container-runtime path with a nix store path pointing to the CSV. Modify libnvidia-container0 to naturally trim off the nix store prefix.
4. Hope for best.

I think this approach would naturally work for Nix-built docker containers as well - the Nix containers would presumably contain the relevant l4t packages by default and so would rely on the devices (which should be in the same places).

j-baker, 2023-08-23:
I have an open pr which is a first stab at this.

PR #172 (2023-12-15): Add initial OCI container support
###### Description of changes

These changes add OCI container support for Tegra devices on NixOS. This is accomplished using NVIDIA's newer `nvidia-ctk` tooling, which uses the CDI spec. Docker does not yet support this spec in any released version, but will have experimental support for it in their next release. Podman has supported this spec for a few years. See changes to README.md for more notes.

Closes #5 
Closes #117 

###### Testing

Tested running a simple cuda program on an orin-agx-devkit0 with the nvcr.io/nvidia/l4t-base:35.3.1 image.
closed 2023-12-15 by danielfullmer