anduril/jetpack-nixos #16: eMMC Flash Support

sphw, 2023-01-17:
I'm interested in trying to add support for generating and flashing an eMMC image to `jetpack-nix`. It seems like it should not be a huge undertaking. Are you guys planning on adding support? And/or would you be welcome to a contribution?

There are a few subtleties that I'm interested in before I go off and start developing. It seems like `./flash.sh` expects a separate `esp.img` and `system.img` when writing to eMMC, but Nix's built in image builder creates a combined image. I suspect that you can just use the built-in L4T ESP image, but I'm not sure if that will break things. Any help or guidance you all could offer would be great.

danielfullmer, 2023-01-17:
So, my goal with this was to have the flashing script just create a system that's ready to boot via UEFI like any other x86/arm UEFI system.  Upstream jetpack actually has additional partitions that it flashes to the QSPI/emmc which I remove (kernel/kernel_b/APP/etc): https://github.com/anduril/jetpack-nixos/blob/a6079bc8edce859ea32cdfeb7498af9b2d0a6395/modules/devices.nix#L39-L45
Their `esp.img` includes an EFI application called L4TLauncher.efi that enables backwards compatibility with the way booting worked on pre-Jetpack 5 devices, e.g. using an extlinux.conf file.  I personally didn't see any need for this and just wanted to stick with standard UEFI booting on NixOS (systemd-boot/grub/etc).  Actually flashing the device with a NixOS system was outside the scope of what I was intending.

While I wasn't planning on adding support for this, but I'd be curious to hear why you'd like this. Is it just to flash everything using one command? We currently have a two-step process for flashing where we first run this flashing script and then boot via usb or the network to partition and install a NixOS system.

Additionally, I believe the Orin NX only has QSPI, no eMMC at all. So I don't believe it'd be possible to flash a full NixOS system using the flashing script.

sphw, 2023-01-17:
> While I wasn't planning on adding support for this, but I'd be curious to hear why you'd like this. Is it just to flash everything using one command? We currently have a two-step process for [...]

It is mainly to support a single step installation. I'd also like to be able to use the Jetson OTA update system to update the entire image at once. Would you still be open to this feature being added, even though it's not your exact target use case?

> Additionally, I believe the Orin NX only has QSPI, no eMMC at all. So I don't believe it'd be possible to flash a full NixOS system using the flashing script
I'm currently targeting the Xavier AGX which has 64GB of eMMC. I believe the new Orin AGX modules have the same.

danielfullmer, 2023-01-18:
Sure, I'd at least take a look at it.  I assume the change would involve modifying the partitionTemplate and passing an image through to the flash.sh script. If it's not too involved and difficult to support going forward, we could merge it.

mikatammi, 2023-05-25:
We have done it the other way around, we generate the esp.img and root.img in our repository and then import the flash-script.nix from jetpack-nixos and provide it the partitionTemplate that has the esp.img and root.img in place.

This commit shows most of the changes to support the currently latest jetpack-nixos (pinning into this commit because we might refactor and move the stuff to different files later):
https://github.com/tiiuae/ghaf/commit/35c9adf64b34b22625b8bdff0b8de950bd7f0eea

ebrensi, 2025-09-15:
Hello!  Just saw this.  I have been successful in creating a custom jetpack-nixos ISO installer for the Orin Nano.  As in the instructions it requires booting from the USB ISO and running an install script.  The script then uses disko to format the SD card (eMMC?), copy over nix store files etc so that NixOS boots from the onboard SD card.  It seems like it should not be too much more trouble to write an image directly to the SD card, but apparently it is?  If I create an sd image of the `nixosSystem` from my flake and try that in the Orin Nano it does not boot, so I guess not.