anduril/jetpack-nixos #29: Flashing UEFI Firmware stuck on Auvidea JNX30D

Shoadi, 2023-02-28:
Hello community,

I am trying to follow the instructions for flashing the UEFI-firmware of anduril/jetpack-nixos on an Auvidea JNX30D - unfortunately without success, as the installer is stuck in an:
`[   7.6925 ] tegrarcm_v2 --isapplet
[ 1022.7312 ] tegrarcm_v2 --ismb2
[ 2038.5395 ] tegradevflash_v2 --iscpubl
[ 2038.5405 ] CPU Bootloader is not running on device.`
- loop.

Is this happening due to a lack of support for the DevKIT, or is this something I could solve quite simple?

Find the full log attached.

Thanks and have a great day.

[log.txt](https://github.com/anduril/jetpack-nixos/files/10849094/log.txt)

danielfullmer, 2023-03-03:
It's possible. Some carrier boards have additional BSP changes to pinmux/device tree/other settings that might not be compatible with the standard xavier-nx-devkit configuration.

But since you're using the Xavier NX, there's one thing you can check first: Is this a developer Xavier NX with an SD card on the SoM, or the a production Xavier NX with the emmc instead of an SD card?  You'll need to use the flashing script that corresponds to the one you have, e.g. `xavier-nx-devkit` for the development module or `xavier-nx-devkit-emmc` for the production module.

Shoadi, 2023-03-07:
Thank you very much for your reply -  Auvidea JN30D DevKITs have a Jetson Xavier NX **production module** assembled - therefore `xavier-nx-devkit-emmc` was the way which lead to a successful flashing of the device.

So regarding BSP changes - I found that Auvidea is providing a ./kernel_out/* folder containing a kernel-image as well as a folder of devicetree-blobs.

Originally using the Nvidia SDK-manager one needs to:

```
cp ~/kernel_out/* /home/USER/nvidia/nvidia_sdk/JetPack_X.X.X_DP_Linux_JETSON_XAVIER_NX_TARGETS/Linux_for_Tegra
cd /home/USER/nvidia/nvidia_sdk/JetPack_X.X.X_Linux_JETSON_XAVIER_NX_TARGETS/Linux_for_Tegra
sudo ./apply_binaries.sh
```
...and then finishing to flash with:

`sudo ./flash jetson-xavier-nx-devkit-emmc mmcblk0p1`

- how can I do these changes for the jetpack-nixos?

danielfullmer, 2023-03-09:
So it depends on what exactly apply_binaries.sh does, but we have a similar carrier board from another vendor that basically copies a bunch of files into the flash tools directory before running the flashing script. You can execute commands that modify the flash tools by setting
```
hardware.nvidia-jetpack.flashScriptOverrides.postPatch = ''
  echo EXAMPLE
''
```

If you copy prebuilt device trees to replace the ones produced by jetpack-nixos you'd lose the ability to modify them, however.  If auvidea provides a kernel patch which includes their modifications, that would be a better way to handle it.

closed 2023-03-18 by danielfullmer