anduril/jetpack-nixos #63: TX2 Support

Wulfsta, 2023-04-19:
This is not a feature request, but I am curious what work would need to be done to add support for the Jetson TX2? I dug one out of storage recently and would love to run NixOS on it.

danielfullmer, 2023-04-20:
We'd have to package Jetpack 4, which has enough differences compared with Jetpack 5 (old kernel, different bootloader) that I opted not to try.  This could possibly be done by adding enough conditionals everywhere to capture the differences between Jetpack 4 and Jetpack 5.  I haven't really looked closely at what would be required for this to happen.

However, when Jetpack 6 is released, it is expected to drop support for Xavier AGX and Xavier NX, and I'd like to keep Xavier AGX/NX support continuing, so at that time, we'll have to have some level of support for multiple versions of Jetpack.

pbsds, 2024-05-13:
https://forums.developer.nvidia.com/t/jetpack-6-0-production-release-announcement/291709
https://docs.nvidia.com/jetson/jetpack/release-notes/index.html
https://jetsonhacks.com/2024/05/06/jetpack-6-production-release/

A year have passed, what is your current support strategy for Xavier? If two jetpack are to be supported, is there any interest in TX1/2?

theotheroracle, 2024-11-08:
i'm interested in helping with tx1 support ! ( i can open a seperate issue for that if desired )