anduril/jetpack-nixos #85: X11 display not working on Orin Nano

danielfullmer, 2023-05-16:
I am able to get NixOS up over the serial console and installed and booting off a microSD card.

But I cannot get X11 to work at all and provide display output. I tried enabling modesetting in the nvidia-jetpack option group and that did not help.

<details>
```
(II) NVIDIA(0): Creating default Display subsection in Screen section
        "Screen-nvidia[0]" for depth/fbbpp 24/32
(==) NVIDIA(0): Depth 24, (==) framebuffer bpp 32
(==) NVIDIA(0): RGB weight 888
(==) NVIDIA(0): Default visual is TrueColor
(==) NVIDIA(0): Using gamma correction (1.0, 1.0, 1.0)
(**) NVIDIA(0): Option "AllowEmptyInitialConfiguration" "true"
(**) NVIDIA(0): Enabling 2D acceleration
(II) Loading sub module "glxserver_nvidia"
(II) LoadModule: "glxserver_nvidia"
(II) Loading /nix/store/vl7jvh572swjsll0hrarb2967jc44lhk-nvidia-l4t-3d-core-35.3.1-20230319081403/lib/xorg/modules/extensions/libglxserver_nvidia.so
(II) Module glxserver_nvidia: vendor="NVIDIA Corporation"
        compiled for 1.6.99.901, module version = 1.0.0
        Module class: X.Org Server Extension
(II) NVIDIA GLX Module  35.3.1  Release Build  (bugfix_main)  (buildbrain@mobile-u64-6289-d7000)  Sun Mar 19 08:02:08 PDT 2023
(II) NVIDIA: The X server supports PRIME Render Offload.
(--) NVIDIA(0): Valid display device(s) on GPU-0 at SoC
(--) NVIDIA(0):     DFP-0
(--) NVIDIA(0):     DFP-1
(WW) NVIDIA: No DRM device: Direct render devices found but none could be
(WW) NVIDIA:     used.
(II) NVIDIA(0): NVIDIA GPU Orin (nvgpu) (GA10B) at SoC (GPU-0)
(--) NVIDIA(0): Memory: 6631800 kBytes
(--) NVIDIA(0): VideoBIOS:
(EE) NVIDIA(GPU-0): Failed to acquire modesetting permission.
(EE) NVIDIA(0): Failing initialization of X screen
(II) UnloadModule: "nvidia"
(II) UnloadSubModule: "glxserver_nvidia"
(II) Unloading glxserver_nvidia
(II) UnloadSubModule: "wfb"
(II) UnloadSubModule: "fb"
(EE) Screen(s) found, but none have a usable configuration.
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/dev/null" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
```
</details>

_Originally posted by @tpwrules in https://github.com/anduril/jetpack-nixos/issues/78#issuecomment-1546491053_

----

Additional note: This also hasn't been tested on the Orin NX, so I think its likely broken there as well.

tpwrules, 2023-07-02:
Has this been tested or addressed?

danielfullmer, 2023-07-02:
I haven't looked into this issue yet.

hacker1024, 2023-07-12:
We can't get any display output on our Orin NXs. As soon as GRUB launches the kernel, the display stops receiving a signal. Booting with `video=efifb:list` shows a reasonable list of display modes, but manually choosing them does not help. `nomodeset` has no effect.

hacker1024, 2023-12-12:
With LightDM and no modesetting, X11 (GNOME) is working on my Orin NX. Thanks!

closed 2023-12-12 by danielfullmer

tpwrules, 2023-12-12:
Confirmed that this works on Orin Nano too.