anduril/jetpack-nixos #110: Jetson Orin NX - black screen after GRUB

hacker1024, 2023-07-18:
Using the minimal ISO on a Jetson Orin NX, I get nothing but a black screen displayed after choosing an option in the GRUB menu. I can access the Jetson through SSH, though - it's just a display issue.

I have tried removing the `video=efifb:off` argument, but this has not helped. It does not cause any crashes, though.

These lines of the kernel log seem relevant:

```
[    0.592687] efifb: probing for efifb
[    0.592724] efifb: framebuffer at 0x440000000, using 8640k, total 8640k
[    0.592726] efifb: mode is 1920x1080x32, linelength=8192, pages=1
[    0.592727] efifb: scrolling: redraw
[    0.592730] efifb: Truecolor: size=8:8:8:8, shift=24:16:8:0
[    0.592851] fbcon: Deferring console take-over
[    0.592854] fb0: EFI VGA frame buffer device
[    0.594377] tegra_dc_assign_hw_data: no matching compatible node
[    0.594406] fbcon: Taking over console
[    0.594410] tegradccommon module_init failed
[    0.594414] tegradc module_init failed
[    0.594542] Console: switching to colour frame buffer device 240x67
```

hacker1024, 2023-07-18:
It looks like the UEFI settings menu also does not display with the firmware capsule built from `#devicePkgs.orin-nx-devkit.uefiCapsuleUpdate`.

yannham, 2023-09-27:
I encounter the same issue on a Jetson AGX. I can't show the firmware menu either. @hacker1024 , how did you get SSH to work? I thought you need to start `sshd` manually even when it's enabled in the installer image.

hacker1024, 2023-09-27:
IIRC sshd runs by default, but there's no password set. We build custom ISO images that have users with passwords.

danielfullmer, 2023-09-27:
I don't believe the EFI Linux console is expected to work on Orin platforms.  However, wayland and X11 should work, so you can get graphical output.  I personally just use the serial interface instead.

closed 2023-09-27 by danielfullmer

heshdotcc, 2024-07-16:
I've also lost the BIOS settings menu after flashing an Orin AGX.

I could boot the minimal ISO with everything working and even successfully install it onto an F2FS root NVMe disk partition.

After that, it reboots and hangs without video. There is no `system-boot` or video for the BIOS menu.

I want to document this, so I appreciate any help!

yannham, 2024-10-17:
@heshdotcc for what it's worth, I used the serial port to set up jetpack-nixos and proper SSH credentials initially and then just used SSH from then on. I've not managed to get the video working at that time.