anduril/jetpack-nixos #111: Orin NX 16gb on Orin Nano DevKit has issues booting via USB on firmware 35.3.1

chansen-and, 2023-07-19:


danielfullmer, 2023-08-18:
We were actually unable to reproduce this issue on our Orin NX or Orin Nano.  I just merged a PR that includes an update to the usb controller firmware that NVIDIA includes, which may help. Would you be able to reflash and test again with firmware after https://github.com/anduril/jetpack-nixos/commit/48d86a9a17730154670b5216c44b629f7d32672c ?

danielfullmer, 2023-09-27:
Just checking in again if you were able to reproduce this issue with the latest commit?

chansen-and, 2023-10-03:
@danielfullmer Yep, I tested it and still am seeing the same issue.

TLDR for those who stumble upon this issue:
- When I have an SSD Plugged in I am able to see USB as a bootable option. 
- When I remove the SSD I no longer see the USB as a bootable option. 

I'll test again with 35.4.1 at some point.

emrahbillur, 2024-01-18:
I'm facing the same issue with Orin Nano Developer Kit, tested with both Orin NX 16 GB and Orin Nano. Same issue faced.
`
Could not locate device tree fixup protocol, skipping.
EFI stub: Booting Linux Kernel...
EFI stub: ERROR: Invalid header detected on UEFI supplied FDT, ignoring ...
EFI stub: Generating empty DTB
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services and installing virtual address map...
`
Tested with even overlays with different device tree configurations for orin nano developer kit. Firmware boots original NVIDIA images from nvme but any nixos image on usb fail.

mikatammi, 2024-04-14:
> I'm facing the same issue with Orin Nano Developer Kit, tested with both Orin NX 16 GB and Orin Nano. Same issue faced.
>
>
> Tested with even overlays with different device tree configurations for orin [...]

This is probably due to fact systemd-boot does not allocate any extra space for the device tree. It can be patched for example like this (a bit ugly): https://github.com/tiiuae/ghaf/blob/main/modules/common/systemd/systemd-boot-double-dtb-buffer-size.patch  Without the modification, because there is not any extra space, the UEFI will fail applying the overlay changes from its .dtbo files

Another way probably could be to modify the .dtb file to have some extra space in end of it