anduril/jetpack-nixos #118: Gnome on Orin AGX and Orin Nano reports memory manager issues and then errors

j-baker, 2023-07-30:
I could use some help here! I've got an Orin AGX and an Orin Nano, both custom carrier boards from different vendors. When they run NixOS and I enable Gnome through the 'gnome' settings, I get errors like this during x startup.

```
Jul 27 12:26:37 nixos /nix/store/jsmfi4447h39f4xrawfhajjdvnqglzqi-gdm-43.0/libexec/gdm-x-session[1085]: (II) Loading /nix/store/vl7jvh572swjsll0hrarb2967jc44lhk-nvidia-l4t-3d-core-3>
Jul 27 12:26:37 nixos /nix/store/jsmfi4447h39f4xrawfhajjdvnqglzqi-gdm-43.0/libexec/gdm-x-session[1085]: NvRmMemInitNvmap failed with Permission denied
Jul 27 12:26:37 nixos /nix/store/jsmfi4447h39f4xrawfhajjdvnqglzqi-gdm-43.0/libexec/gdm-x-session[1085]: 549: Memory Manager Not supported
Jul 27 12:26:37 nixos /nix/store/jsmfi4447h39f4xrawfhajjdvnqglzqi-gdm-43.0/libexec/gdm-x-session[1085]: ****NvRmMemInit failed**** error type: 196626
Jul 27 12:26:37 nixos /nix/store/jsmfi4447h39f4xrawfhajjdvnqglzqi-gdm-43.0/libexec/gdm-x-session[1085]: *** NvRmMemInit failed NvRmMemConstructor
```

X then crashes with a segfault. I don't think this is actually a permissions issue - rather I think something has exploded in a place nVidia does not expect. If I try KDE, I get no errors (unless modesetting is on) but there are no displays - no framebuffers registered and the HDMI port does not work. I'm wondering if these two are related somehow, but not sure.

Do you have any ideas how one would go about debugging this? I'm wondering if it's missing a library or kernel module or something which is stopping things from working nicely.

maleadt, 2024-01-25:
I just ran into this as well on an Xavier AGX, turns out it _can_ be permission related, and the user needs to be part of the `video` group (https://forums.developer.nvidia.com/t/jetpack-5-0-dp-nvrmmeminitnvmap-failed-with-permission-denied/219804).

mankicks, 2026-01-07:
I encountered this issue on the Tegra `Ubuntu 24.04.3 LTS` kernel=`GNU/Linux 6.8.12-tegra aarch64` on AGX Thor when using podman (rootless). I gave my podman user all the necessary groups including `video` to no avail. The setting `no-cgroups = true` (under first stanza `[nvidia-container-cli]` in `/etc/nvidia-container-runtime/config.toml`) has no effect on podman. Obviously there are many other workarounds and methods to disable cgroups/namespace isolation, permanently or temporarily as one might need, but that undermines a key benefit of rootless podman.

`/etc/udev/rules.d/99-tegra-devices.rules` line `KERNEL=="nvmap" OWNER="root" GROUP="video" MODE="0440"` : Changing `0440` to `0444` (granting read to 'other') allows global read and resolves the issue. Mapping the `video` group to allow "impersonation" by subuid as described [here](https://github.com/containers/podman/blob/main/docs/tutorials/rootless_tutorial.md) would probably be much safer. Vulnerabilities relating to nvmap have been disclosed, too, so might want to check those out before deciding how to proceed...I will come back to this later (probably after someone hacks my Jetson).

Hope this helps someone. Cheers!