anduril/jetpack-nixos #131: segfault in multimedia samples

freeman94, 2023-09-21:
I am trying to run the `video_decode` sample in `samples.multimedia-samples` and getting a segfault. I've verified that the same works on a board running Ubuntu with the latest jetpack version.

Device info:
```
$ ota-check-firmware
Current firmware version is: 35.3.1
Current software version is: 35.3.1

$ hostnamectl
 ...
       Icon name: computer
      Machine ID: 68d20dd1d0984bcbb7bbac72a41c6e93
         Boot ID: 5317686e6c4f466384c82d862198954f
Operating System: NixOS 23.11 (Tapir)
          Kernel: Linux 5.10.104
    Architecture: arm64
 Hardware Vendor: Unknown
  Hardware Model: NVIDIA Orin NX Developer Kit
Firmware Version: 35.3.1
   Firmware Date: Tue 1980-01-01
```
Stack trace:
```
$ nix build github:anduril/jetpack-nixos#legacyPackages.aarch64-linux.samples.multimedia-samples
$ sudo gdb ./result/bin/video_decode
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-unknown-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./result/bin/video_decode...
(No debugging symbols found in ./result/bin/video_decode)
(gdb) r H264 output.mkv --disable-rendering
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
        info "(gdb)Auto-loading safe path"
warning: File "/nix/store/r0aclnvj3xz61pqvdbd4l57z714ybywa-glibc-2.35-224/lib/libthread_db.so.1" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load:/nix/store/cq3kvw7kwz2n9c13fg6qaaj23mkm116l-gcc-12.3.0-lib".
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be availa
ble.
[New LWP 1166]
Creating decoder in blocking mode
Opening in BLOCKING MODE
NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
[New LWP 1167]
[New LWP 1168]
[New LWP 1169]
NvMMLiteBlockCreate : Block : BlockType = 261
Setting frame input mode to 1
[New LWP 1170]
Input file read complete
[New LWP 1171]
Starting decoder capture loop thread
Got EoS at output plane
[ERROR] (NvV4l2Element.cpp:114) <dec0> Error while DQing event: Invalid argument
Error in dequeueing decoder event
Exiting decoder capture loop thread
[LWP 1171 exited]
[LWP 1170 exited]

Thread 1 "DecOutPlane" received signal SIGSEGV, Segmentation fault.
0x0000fffff487d604 in __pthread_clockjoin_ex ()
   from /nix/store/r0aclnvj3xz61pqvdbd4l57z714ybywa-glibc-2.35-224/lib/libc.so.6
(gdb) r H264 output.mkv bt
The program being debugged has been started already.
Start it from the beginning? (y or n) Quit
(gdb) bt
#0  0x0000fffff487d604 in __pthread_clockjoin_ex ()
   from /nix/store/r0aclnvj3xz61pqvdbd4l57z714ybywa-glibc-2.35-224/lib/libc.so.6
#1  0x000000000043d24c in NvV4l2ElementPlane::waitForDQThread(unsigned int) ()
#2  0x000000000043da60 in NvV4l2ElementPlane::deinitPlane() ()
#3  0x000000000042b880 in NvV4l2Element::~NvV4l2Element() ()
#4  0x000000000043e484 in NvVideoDecoder::~NvVideoDecoder() ()
#5  0x0000000000406be8 in main ()
```

danielfullmer, 2023-09-21:
I'm not able to reproduce the failure in the same way you're seeing.

When I run it, it fails when trying to render to the display, likely because I don't have X11/wayland running on this device)
```
$ /nix/store/m010m4has2b691fwzpdp212c42hxzz13-multimedia-samples-35.3.1-20230319081403/bin/video_decode H264 sample_outdoor_car_1080p_10fps.h264 
Creating decoder in blocking mode 
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Setting frame input mode to 1 
Starting decoder capture loop thread
Video Resolution: 1920x1080
Video SAR width: 1 SAR height: 1
[ERROR] (NvEglRenderer.cpp:98) <renderer0> Error in opening display
[ERROR] (NvEglRenderer.cpp:154) <renderer0> Got ERROR closing display
Segmentation fault (core dumped)
```
One possible difference, is I tested on 23.05, not nixpkgs master.

freeman94, 2023-09-21:
Try running with the `--disable-rendering` flag.

danielfullmer, 2023-09-21:
Ah, nice. I didn't notice that flag. In that case, yes, I can reproduce the segfault.  In my case, it still seems to get slightly further and output these lines:
```
Video Resolution: 1920x1080
Video SAR width: 1 SAR height: 1
```
but it then fails with a very similar backtrace.

freeman94, 2023-09-21:
Beginning to think this may be a red herring. I have carefully gone through the samples to go h264 -> YUV -> h264 as expected (using `decode_sample`/`encode_sample`). I came across this originally after seeing the same segfault in (non-nvidia) vendor code. I will try other options and report back.

commit a148358e (2023-09-21): Add a patch to fix video_decode multimedia sample package

Kudos to OE4T for already having a fix for this :)

Tested working on Orin AGX devkit

Fixes #131
closed 2023-09-21 by danielfullmer