anduril/jetpack-nixos #142: Integrate with nixos-generators

bugeats, 2023-10-10:
How cool would it be to see `jetpack` in the [list of nixos-generators](https://github.com/nix-community/nixos-generators#supported-formats).

mikatammi, 2023-10-25:
We actually wrote "simple" nixos-generator for jetpack-nixos into our project, the entry point is here: https://github.com/tiiuae/ghaf/blob/main/modules/hardware/nvidia-jetson-orin/format-module.nix

We use sd-image as a basis because it works with cross-compilation, and then manually creates the structure into FAT partition that is EFI/ESP compatible and seems to work with NVIDIA's edk2-bootloader.

There's also example of how to generate flash script which also flashes the root filesystem:
https://github.com/tiiuae/ghaf/blob/main/lib/mk-flash-script/default.nix

It also involves module which generates the partition template here:
https://github.com/tiiuae/ghaf/blob/main/modules/hardware/nvidia-jetson-orin/partition-template.nix

danielfullmer, 2023-11-17:
I'm not currently planning to implement something specific to jetsons here in this repo. The goal is to make it as close to a standard UEFI ARM device and then whatever mechanism you decide for installing a NixOS system on a UEFI system should be able to work on jetsons.

(Since we now have an "initrd"-based flashing method, it may be possible to allow the user to do additional things in that script after flashing the firmware, like writing out a NixOS system to an nvme drive--but that is all speculative.)

closed 2023-11-17 by danielfullmer