anduril/jetpack-nixos #153: Usage of OpenSSL 1.1 causes flashing script builds to fail against new versions of NixOS

mikatammi, 2023-11-10:
After this commit https://github.com/anduril/jetpack-nixos/commit/a829d807e1030753f81d83674e52cfa7cadc742e flash script builds start to fail with following error, if built against latest NixOS 23.05 and nixos-unstable channel versions.

There does not seem to be easy and clean way to work around this. All workarounds I've tried involve building stuff with `--impure` flag. Basically only a and c work, b does not work at all.

My question is, is OpenSSL 1.1 strictly required, or could we somehow patch our way around it. Does fuse and secure boot related scripting need legacy OpenSSL?

```
error: Package ‘openssl-1.1.1w’ in /nix/store/3s69yxbbl116zwga3i6cy7prplywq0bn-source/pkgs/development/libraries/openssl/default.nix:210 is marked as insecure, refusing to evaluate.

       Known issues:
        - OpenSSL 1.1 is reaching its end of life on 2023/09/11 and cannot be supported through the NixOS 23.05 release cycle. https://www.openssl.org/blog/blog/2023/03/28/1.1.1-EOL/

       You can install it anyway by allowing this package, using the
       following methods:

       a) To temporarily allow all insecure packages, you can use an environment
          variable for a single invocation of the nix tools:

            $ export NIXPKGS_ALLOW_INSECURE=1

        Note: For `nix shell`, `nix build`, `nix develop` or any other Nix 2.4+
        (Flake) command, `--impure` must be passed in order to read this
        environment variable.

       b) for `nixos-rebuild` you can add ‘openssl-1.1.1w’ to
          `nixpkgs.config.permittedInsecurePackages` in the configuration.nix,
          like so:

            {
              nixpkgs.config.permittedInsecurePackages = [
                "openssl-1.1.1w"
              ];
            }

       c) For `nix-env`, `nix-build`, `nix-shell` or any other Nix command you can add
          ‘openssl-1.1.1w’ to `permittedInsecurePackages` in
          ~/.config/nixpkgs/config.nix, like so:

            {
              permittedInsecurePackages = [
                "openssl-1.1.1w"
              ];
            }
(use '--show-trace' to show detailed location information)
```

danielfullmer, 2023-11-13:
Apologies.  I should have tested this against 23.05.  We have something that (unfortunately), still requires OpenSSL 1.1.x--but we can workaround the issue outside of jetpack-nixos.

closed 2023-11-13 by danielfullmer

mikatammi, 2023-11-14:
I'm just about to start playing around with fuse scripts and secure boot, so I would be interested does this affect the functionality of fuse scripts?

danielfullmer, 2023-11-15:
Not as far as I am aware. I've tested the fusing / secure boot workflow on Xavier AGX and Orin NX.  It's still not well documented, and takes some work to generate all the right keyfiles / fusing xml files, but it does work.

PR #154 (2023-11-13): Revert "Switch flash scripts to use from openssl to openssl_1_1"
This reverts commit a829d807e1030753f81d83674e52cfa7cadc742e

Closes #153