anduril/jetpack-nixos #155: Support for Nvidia DRIVE Orin

mode89, 2023-11-14:
Hi :)

Would it work on an [Nvidia DRIVE Orin](https://www.nvidia.com/en-us/self-driving-cars/hardware/) board?

danielfullmer, 2023-11-17:
I don't know! Do you have one? :)

mode89, 2023-11-17:
OK :)

I may get one soon. Would like to try it there. Do you see any configuration parts that may require special attention?

By the way, would it possible to flash it back with the original firmware, if Nix wouldn't work?

danielfullmer, 2023-11-17:
As far as I can tell, the devkit and software package for DRIVE are not publicly available, so I can't confidently say how different it would be from standard Jetson software/firmware.   I'd guess that you should be able to reflash it back to original firmware/software--but it's just a guess.

mode89, 2023-11-17:
I see. Thanks!

Sorry, I forgot to mention, that I wasn't expecting to hear anything but just speculations :)

Is there such problem on Jetson? Does nix conversion go very deep into the firmware.

danielfullmer, 2023-11-21:
There aren't too extensive changes to the firmware in this repository, unless you're doing things like enabling secure boot.  I'd expect the unmodified firmware may work (if it's new enough that it supports generic UEFI booting)

danielfullmer, 2023-11-21:
Just for completeness: I guess I should say that this isn't a feature we're currently targeting, and even if we did, depending on licensing, there is a good chance we couldn't release it publicly.