anduril/jetpack-nixos #184: Runtime PTX compilation not supported

maleadt, 2024-01-25:
Compiling PTX code dynamically, i.e., by calling `cuModuleLoad` in `libcuda`, doesn't work on a system with `jetpack.l4t-cuda` installed: It errors with `ERROR_JIT_COMPILER_NOT_FOUND`, which I guess indicates that some dynamically-loaded libraries are missing. I don't have another Tegra system to check, but on x86 at least `libcuda.so` is bundled together with `libnvidia-ptxjitcompiler`, `libnvidia-nvvm` and `libcudadebugger`.

danielfullmer, 2024-03-27:
Thanks for reporting this, and apologies for taking so long to respond. Do you have a example I could use to reproduce this issue?

maleadt, 2024-03-27:
Here's a C reproducer that I confirmed works only when adding l4t-3d-core do my list of packages:

```c
#include <stdio.h>
#include <cuda.h>

const char *dummy_ptx =
".version 3.1\n"
".target sm_20\n"
".address_size 64\n"
".visible .entry dummy() {\n"
"  ret;\n"
"}";

int main() {
    CUresult res;
    CUdevice cuDevice = 0;
    CUcontext cuContext = 0;
    CUmodule cuModule = 0;

    res = cuInit(0);
    if (res != CUDA_SUCCESS) {
        printf("cuInit failed: res = %d\n", res);
        return -1;
    }

    res = cuDeviceGet(&cuDevice, 0);
    if (res != CUDA_SUCCESS) {
        printf("cuDeviceGet failed: res = %d\n", res);
        return -1;
    }

    res = cuCtxCreate(&cuContext, 0, cuDevice);
    if (res != CUDA_SUCCESS) {
        printf("cuCtxCreate failed: res = %d\n", res);
        return -1;
    }

    res = cuModuleLoadData(&cuModule, dummy_ptx);
    if (res != CUDA_SUCCESS) {
        printf("cuModuleLoadData failed: res = %d\n", res);
        return -1;
    }

    cuModuleUnload(cuModule);
    cuCtxDestroy(cuContext);

    return 0;
}
```

PR #234 (2024-08-20): Move libnvidia-ptxjitcompiler library into l4t-cuda
###### Description of changes

This library is needed for runtime PTX compilation. It's unclear why NVIDIA has this library in l4t-3d-core and not in l4t-cuda. Even more so since the cuda compat package has libcuda as well as libnvidia-ptxjitcompiler in the same package. meta-tegra does a similar thing where they pull libnvidia-ptxjitcompiler out of l4t-3d-core and place it in the same package as libcuda.

Additionally, add matrixMul* samples, including matrixMulDynlinkJIT specifically, which was failing to run previously.

Fixes #184 

##### [... cut]
commit 794c2aa1 (2024-08-20): Move libnvidia-ptxjitcompiler library into l4t-cuda

This library is needed for runtime PTX compilation. It's unclear why
NVIDIA has this library in l4t-3d-core and not in l4t-cuda. Even more so
since the cuda compat package has libcuda as well as
libnvidia-ptxjitcompiler in the same package. meta-tegra does a similar
thing where they pull libnvidia-ptxjitcompiler out of l4t-3d-core and
place it in the same package as libcuda.

Additionally, add matrixMul* samples, including matrixMulDynlinkJIT
specifically, which was failing to run previously.

Fixes #184
closed 2024-08-20 by danielfullmer