anduril/jetpack-nixos #217: Jetpack 6 Support

adnanwahab, 2024-05-22:
Hi, i found this repo to be very useful for installing nixos on my jetson.

I was wondering if there was any plans to support Nvidia Jetpack 6? https://developer.nvidia.com/embedded/jetpack

Or if not, would you have any pointers as to how i could get started updating this repo and making a PR ?

Thank you

vadika, 2024-05-29:
We've forked and now working on porting latest Jetpack with near mainline kernel. 
WIP repository is https://github.com/tiiuae/jetpack-nixos/

jemaw, 2024-07-03:
Do you plan on upstreaming the changes into this repo here?

vadika, 2024-07-12:
> Do you plan on upstreaming the changes into this repo here?

Well, we'd like to, but not sure if will be welcomed by anduril. Anyway PR (WIP) is in place.

kihehs, 2024-08-30:
I don't see a PR. Did it get closed?

vadika, 2024-10-31:
PR now is ready for checking and testing -- would appreciate this.
https://github.com/anduril/jetpack-nixos/pull/224

dran1x, 2025-01-13:
Any updates on this?

elliotberman, 2025-07-01:
Closed by #291

closed 2025-07-01 by elliotberman