anduril/jetpack-nixos #226: SoC-specific firmware cannot be found

hacker1024, 2024-07-10:
Some SoC-specific firmware (`lib/firmware/tegra*`) in the `l4t-firmware` package cannot be found by the kernel, despite `l4t-firmware` being in the NixOS firmware package list.

For example, the following error shows up in the kernel log when attempting to use NVENC:

```
falcon 154c0000.nvenc: Direct firmware load for nvhost_nvenc080.fw failed with error -2
falcon 154c0000.nvenc: looking for firmware in subdirectory
(NULL device *): nvhost_channelctl: invalid cmd 0x80685600
```

I suspect the firmware files need to be moved out of their `tegraXXX` directory and into the main tree. This can be done based on the given model in the NixOS configuration.

danielfullmer, 2024-07-10:
I'm a bit surprised I haven't run into this myself, since we have a test that (I believe) uses nvenc.
If you run `nix run github:anduril/jetpack-nixos#samples.multimedia-test` , do you get this same kernel error message?

hacker1024, 2024-07-10:
My bad; it appears that the kernel attempts to [load firmware from the main directory first before then loading it from the correct subdirectory](https://github.com/OE4T/linux-tegra-5.10/blob/oe4t-patches-l4t-35.5.0/nvidia/drivers/video/tegra/host/bus_client.c#L1814-L1842), which is succeeding.

I haven't yet been able to test NVENC properly due to what I suspect is [this bug](https://github.com/OE4T/meta-tegra/issues/785). I'm looking at using OE4T's shim now.

closed 2024-07-10 by hacker1024

hacker1024, 2024-07-11:
More details in https://github.com/anduril/jetpack-nixos/pull/228.