anduril/jetpack-nixos #349: Orin NX/ AGX caught into boot loop if HDMI/DP connected

TanelDettenborn, 2025-08-29:
Orin NX and AGX is caught into boot loop if HDMI/DP cable is connected.

**Reproduce steps**
Using Orin NX as an example.

_Device preparation_
Connect display to carrier board via DP/HDMI cable and \*power on\* display.

_Source code_
git clone https://github.com/anduril/jetpack-nixos.git
git rev-parse HEAD
(^^ prints d9530d50b82b71c224046ce95c04d06cb218f297)

_Build_
nix build .#flash-orin-nx-devkit
nix build .#iso_minimal

_Flash and dd_
sudo ./result/bin/flash-orin-nx-devkit
sudo dd if=./result/iso/nixos-minimal-25.05.20250804.ce01dae-aarch64-linux.iso of=/dev/sdX bs=1M oflag=sync status=progress

_Logs_
Below are the last line of UART (before device reboot itself).
```
Jetson System firmware version 36.4.4-0978d5e6ec30 date 1980-01-01T00:00:00+00:0                                                                                                                            
0                                                                                                                                                                                                           
ESC   to enter Setup.                                                                                                                                                                                       
F11   to enter Boot Manager Menu.                                                                                                                                                                           
Enter to continue boot.                                                                                                                                                                                     
                                                                                                                                                                                                            
Loading graphical boot menu...                                                                                                                                                                              
                                                                                                                                                                                                            
Press 't' to use the text boot menu on this console...                                                                                                                                                      
                                                                                                                                                                                                            
                                                                                                                                                                                                            
                                                                                                                                                                                                            
Synchronous Exception at 0x000000045E1DE8E0                                                                                                                                                                 
                                                                                                                                                                                                            
                                                                                                                                                                                                            
Synchronous Exception at 0x000000045E1DE8E0                                                                                                                                                                 
ASSERT [ArmCpuDxe] /build/edk2/ArmPkg/Library/DefaultExceptionHandlerLib/AArch64/DefaultExceptionHandler.c(345): ((BOOLEAN)(0==1))                                                                          
                                                                                                                                                                                                            
Resetting the system in 5 seconds.                                                                                                                                                                          
Shutdown state requested 1                                                                                                                                                                                  
Rebooting system ...                                                                                                                                                                                        
                       
```

danielfullmer, 2025-10-05:
I don't encounter this issue on my Orin AGX devkit. Is the issue still present with the latest revision on master? You said you have the same issue on both Orin AGX and NX. Could you try connecting to another monitor?

TanelDettenborn, 2025-10-07:
I should have been more thorough with tests and I should have tested Orin devices with multiple monitors. I was not considering it, because the jetpack-nixos have worked fine with current _monitor_. Monitor in italic, because I have not used a dedicated physical monitor. I have should have mentioned this in reproduction steps!!

The issues is still present in the current master (tested Orin NX/AGX devkits), but it is only present within monitor's "small PIP". As mentioned above then I have not used a dedicated monitor for displaying Orin device output, but I have used my monitor's Picture In Picture (PIP) feature. For PIP it supports two different PIP sizes: small and large.

For the sake of testing I have tested jetpack-nixos with 7" and 34" monitors and with both monitors jetpack-nixos boots fine. I have also tested jetpack-nixos with my monitors Picture by Picture (PBP) and larger PIP and, again, jetpack-nixos seems to boot fine. I am only able to reproduce it with small PIP.

I am able to prevent jetpack-nixos boot loop by commenting away [NVIDIA_DISPLAY-config from edk2 config](https://github.com/NVIDIA/edk2-nvidia/blob/r36.4.4-updates/Platform/NVIDIA/Jetson/Jetson.defconfig#L20). Previous is a workaround, because nothing is coming to monitor. No NVIDIA splash screen nor graphical boot manager. Monitor stays grey.